T-Troll/alienfx-tools #506: AlienFX Control not recognising supported device

ahrysicos, 2025-03-15:
I have an Alienware AW3423DW monitor. In the past, I have had no issues adding the device to AlienFX Control and customising settings under each tab (seen in tutorial screenshots). It seems that AlienFX Control no longer has a section to add devices and all previously available tabs to customise settings for devices are no longer available. I've tried rolling back to older versions v7 and v8, but same issue, so I guess it's not version dependent?

Also, AlienFX Control no longer starts up with Windows when setting is enabled.

[image]

T-Troll, 2025-03-15:
Wow! It's not detected...

Please check:
- Did USB cable connected to monitor (it used for light control)?
- Did device present into Device manager? What VID/PID it has?
- Any firmware update?

About auto-start - try to uncheck it in Settings and check again. It will request Administrator, accept.

ahrysicos, 2025-03-15:
The monitor doesn't require connection via USB. It is present in Device Manager. No idea about VID/PID. No firmware updates available. I tried unchecking/checking the auto-start setting, It makes no difference. Thanks in advance.

T-Troll, 2025-03-15:
> The monitor doesn't require connection via USB.

... For use it as a monitor. But it required to control lights. So connect it.

> It makes no difference.

Please check Task Scheduler. Did it have "AlienFX GUI" entry?

ahrysicos, 2025-03-16:
Monitor is connected as it should be. Still no luck :(

It seems that unchecking and checking "Start with Windows" has now fixed that issue at least though.