T-Troll/alienfx-tools #508: Fan RPM not showing in 9.1.4.2 (Rework and fixes)

ArisYuuki, 2025-04-06:
**Describe the bug**
During normal usage under both AlienFanGUI and AlienFXGUI, I can set fans just fine, all profiles work as intended, but the RPMS are not showing. I can however get a quick look at RPMs when I click on a lower fan profile. As RPMs drop, you can see the RPMs showing, then back to 0 RPM. 

All profiles do it. This is not an app breaking bug, but more of a small glitch which was not shown in the last version used (9.1.4.2 (Rework and fixes) is the one that has this bug)

WhZzi, 2025-04-10:
I am same problem with you.

 Alienware X15R2

T-Troll, 2025-04-13:
Interesting issue, i'll check this.
Did `alienfan-cli rpm` shows it correctly?
Did increasing "polling time" in Settings solve this issue?

ArisYuuki, 2025-04-13:
Unfortunately Alienfan-cli also didn't show correctly. I did mess around with polling settings but I found the only resolution was to roll back to the prior version.

Other then this small issue and personally having next to no luck with light control (probably user error) this is an amazing app and honestly is greatly appreciated.

I think I should mention I'm using an Alienware M17R4 with an RTX 3080. Not sure if it's specific to this model or not, but that may help you

T-Troll, 2025-04-13:
I check the changes into 4.2 and wonder.

No changes in RPM processing, only some with G-mode detection, which (theoretically) can explain why boost not set.

Can you share me `alienfan-cli rpm` output (i'm interested into your SystemID)?