T-Troll/alienfx-tools #510: Alienware M17 R4 light issue with AlienFX Control

Tanguydruard, 2025-04-12:
Hello, having AWCC that died yesterday without any reason, I searched for for an alternative & I downloaded your app.

I'm using only AlienFX, while i'm not an english expert I had some issues at first about fans and stuff but I could make it work without issue.

Yesterday I used some lights settings for my keyboard, where I could actually preview the correct places with the grids. Actually, I used "load settings" and I think it didn't work ? Or it bugged ? It imported 4 places of keyboard but the preview lights would never turn on, instead it would be light 5 6 7 8 that would fit perfectly with the preview (which didn't appear with "KB left, right etc...")

This morning when I turned on the computer, no more lights & actually when I try to preview now, only the most left side of the keyboard will appear but others won't ever turn green.

I tried to reset & unistall with your file, which didn't work and I am stuck with this for now which is "load settings"  :

[image]

Feel free to ask for any stuff I might have missed !

T-Troll, 2025-04-13:
Seems like you apply configuration from the different model, it's not so good!
For "Detect devices", ALWAYS select EXACT model like yours, not "similar"!

In this case, you may need to clear configuration (using `alienfx-config`) tool and set it up from scratch (or remove all lights and then manually detect it).

"Save settings"/"Load settings" buttons are for light configuration backup/restore only.

Tanguydruard, 2025-04-13:
Okay ! I did it and it worked so far, that's great thank you !

Now another issue... the "Fans&power" dissapeared from my AlienFX window, do you have any idea ? ^^

[image]