T-Troll/alienfx-tools #512: Alienware x17 R2 - alienfan-gui (from 9.1.4) is unusable

chiboul, 2025-04-22:
**Describe the bug**
Starting from version 9.1.4 alienfan-gui became unusable on my x17 R2 as it introduces a critical bug on the machine. Opening the program _seems to_ automatically change the active fan profile every single second, causing constant stuttering and occasional approx. 5 to 8 seconds long stuttering. The fan make a reset noise every second. Besides, the F1 key lights itself at a 1 Hz frequency, confirming that a profile change occurs.
Opening the program on version 9.1.3.1 an below does not do this.

**To Reproduce**
Steps to reproduce the behavior:
1. Open alienfan-gui (from 9.1.4)

**Expected behavior**
Constant 1 Hz stuttering and occassional severe stuttering (approx. 5 to 8 seconds)

**System (please complete the following information):**
 - Model: Alienware x17 R2
 - Model details: [e.g. per-key rgb keyboard]
 - OS: Windows 11 stable

**Additional context**
I've tried every release from alienfan-gui but this bug seems to be persistent across all of them. Furthermore, while the stuttering occurs, I've tried unticking/ticking several options under the Settings tab, but nothing can stop the stuttering/profile change. 
I did not try on latest alienfx-gui because opening this program turns off my keyboard lights completely (it resets the per-key layout I created a few months ago on AWCC, and I don't have AWCC to make it again...)

T-Troll, 2025-04-23:
First, try 9.1.4.3 - i change power mode switching procedure, it should not switch via G-mode anymore.

But this behaviour is not good in general - this means your BIOS report incorrect power mode, so my tools tying to switch it to the chosen every poll. We can diagnose this issue using CLI more precisely.

Your light issue make me sad a bit - it'll only take 5-10 minutes to map your lights (and share your map to help the community), but you prefer to mess with AWCC install/uninstall instead. Ok, the choice is yours. BTW, you can use hardware device effects without lights mapped (Profiles->Device effects).

chiboul, 2025-04-23:
Hi

I've also tried the latest version you uploaded (9.1.4.3) but the issue also happens on this one.
What do you suggest to do with CLI-fan ? I've never used it before.

As for AlienFX, I understand your pov about mapping the keyboard with your program, but I don't find any use case with the hardware effects and macros. I successfully utilized your AlienFX program for years without having to remap the whole thing, but since I updated a few months ago, it started doing this blank lighting overwrite. Therefore I have chosent not to use it anymore for now, and found my self comfortable using only the fan control program.

T-Troll, 2025-04-26:
Ok.
First, open Administrator CMD/Powershell and change a path to AlienFX folder.
Then let's try to check your BIOS readings:
1. `alienfan-cli unlock getpower` - this shows power mode readings after switch to Manual mode
2. `alienfan-cli setpower=1 getpower` - same for non-manual mode
3. `alienfan-cli setgmode=1 gmode getpower` - G-mode check
4. `alienfan-cli setgmode=0 gmode getpower` - Same after g-mode off

Please share the output.

So it's a bug for your lights' configuration, turning it black from time to time? This can be a result of some effect active (f.e. Ambient or Haptics), but colors not selected for it. Please run `alienfx-config backup` from CMD (can be non-Administrator) and share .reg files it creates (config backup), i'll take a look.

chiboul, 2025-04-26:
Hello,

Thanks for your feedback. I've done all the CMD commands and made the alienfx-config backup.
You'll find the screenshot and all the reg files attached here.

[image]

[afx-reg-backup.zip](https://github.com/user-attachments/files/19922809/afx-reg-backup.zip)

To give you more context about the keyboard turning black issue:
My keyboard immediately turns black whenever I open AlienFX-gui. It started doing this after updating to a new afx version a few weeks/months ago. I'm 90% sure I did not press any button in the light effects section because I never used it.
I used to launch AlienFX-gui every day without having this issue. Therefore, I've been only using AlienFan-gui now.

Thanks for your help!

T-Troll, 2025-05-03:
Mystery:
```
C:\…evStudio\alienfx-tools\x64\Release>alienfan-cli.exe setgmode=0             
AlienFan-CLI v9.1.4.3
Supported hardware (5600) detected, 4 fans, 11 sensors, 6 power states, G-Mode, TCC, XMP.                                                                       G-mode Off
```
Maybe some typo in "setgmode"?
Aha... So it is. BIOS report incorrect G-mode status... I make some patches for 4800 in latest release, but seems like not enough - still report incorrect state.
I hope you have AWCC/my GUI quite before this (or it can change state)?

E-Allie, 2025-05-06:
Hello!

Just wanted to report a very similar error to this:

Error description is nearly identical, machine is an m17 R2

For me, last usable version is 9.1.4.2. Every version newer causes this bug

If there's any troubleshooting logs you would want/code you would like executed, feel free to ask!

chiboul, 2025-05-07:
> Maybe some typo in "setgmode"? Aha... So it is. BIOS report incorrect G-mode status... I make some patches for 4800 in latest release, but seems like not enough - still report incorrect state. I hope you have AWCC/my GUI quite before this (or it can change state)?

Hello,
I copied and pasted your lines so I'm unsure where would that typo be.

AWCC is not even installed on my computer, and I can confirm that the AlienFan-gui was not opened during these tests.

T-Troll, 2025-05-09:
@E-Allie, Your case looks similar, but different (according to what i change in 4.3).

2all: In general, both cases are about the fact BIOS report incorrect active mode, so my app trying to change it every time.

Here are debug version with some logging enabled. You will need DbgView utility from Microsoft to capture its log.
Run DbgView, then this debug version, make it running for a while and share the log for me.
[alienfx-gui.zip](https://github.com/user-attachments/files/20116181/alienfx-gui.zip)

T-Troll, 2025-05-25:
Here we are:
```
00000037	14.10000038	[4660] Mon: BIOS mode 7, current 2	
00000038	14.12542057	[4660] Mon: Power mode switch from G-mode to 2	
00000039	14.63557243	[4660] Same profile "Base", skipping switch.	
00000040	14.90232277	[4660] Mon: BIOS mode 7, current 2	
00000041	14.93301773	[4660] Mon: Power mode switch from G-mode to 2	
00000042	15.71662998	[4660] Mon: BIOS mode 7, current 2	
00000043	15.74669933	[4660] Mon: Power mode switch from G-mode to 2	
00000044	16.53274155	[4660] Mon: BIOS mode 7, current 2	
00000045	16.55938721	[4660] Mon: Power mode switch from G-mode to 2
```
As i expect, your BIOS report incorrect power mode (G-mode always reported on at start, then it stuck to Performance power mode).

> Also, my keyboard turn off completly when opening alienfx (I don't have AWCC installed).
It's because you have lights off in settings, turn it on:
```
00000024	14.03265095	[4660] Status: Forced, 0, 0	
00000025	14.03277779	[4660] State update: PID: 47804, brightness: 0, power: 1	
00000026	14.04698372	[4660] State update: PID: 1360, brightness: 0, power: 1
```

About power mode - let me think a bit how to turn off mode 7 in your case... I may need your help with CLI for check.

T-Troll, 2025-05-25:
Oh... Seems like i found the source of this issue.
Please check this debug version - it should fix this issue: 

[alienfx-gui.zip](https://github.com/user-attachments/files/20429651/alienfx-gui.zip)

T-Troll, 2025-05-27:
As i see in logs, you just have lights muted (this can be after "lights off" setting or high dimming (and dimming on) or incorrect screen state (try to uncheck "follow screen state").
I see no issue with lights into this case, except you configuration. You can share it with me (use `aliefx-config` tool), so i can point exact reason.

chiboul, 2025-06-02:
Hello!
I was off this week so I was not able to reach back, sorry for the delay.
I checked every tab on alienfx and every parameters seem fine, I can't figure out what could be causing the keyboard blackout.

I've attached the cmd file into a zip archive. Tell me if I should do anything else

[alienfx-config.zip](https://github.com/user-attachments/files/20547420/alienfx-config.zip)

Thank you!

T-Troll, 2025-06-02:
> I've attached the cmd file into a zip archive

No. You should open CMD into app folder, then RUN it - `alienfx-config backup` - it creates up to 4 .reg files. I need it.

chiboul, 2025-06-02:
Sure, I ran the command and zipped the 4 reg files :

[250602_alienfx-config-backup.zip](https://github.com/user-attachments/files/20547924/250602_alienfx-config-backup.zip)

T-Troll, 2025-06-03:
Yes, mystery a bit - lights not turned off in settings. Maybe a strange bug? I'll check using your settings.

Except one thing - seems like you have Alienware monitor. Then the app starts, it will be switched to control mode, this will turn off lights.

Did you try to set lights to color in Colors tab? Does it work?

chiboul, 2025-06-09:
Hey!
Thanks for looking into this for me.
I'll probably stick to using alienfan-gui for now as I'm not willing to take the time to remap the whole keyboard lighting for now.
That's not an issue because alienfan-gui provides everything I need to manage my fans!

Also, I was wondering if any of your latest releases (9.2.0 and others) introduce a fix for the freeze/stutter issue we first talked about above? I'm down to do some more testing if you want me to. For now I'm sticking with 9.1.3.1 which works fine despite still having a random freeze one every 5 or 6 times I change the fan profile.

Have a great day

chiboul, 2025-06-10:
Hello
I installed 9.2.0 today and I can confirm I don't have the constant 1 Hz stuttering described in my initial post, great news!
While changing the profiles multiple times, I only encountered one "hard throttle" that lasted about 4 to 5 seconds when changing profile from "silent" to "cold". Is this something you want me to log so you can take a look at this behavior?

Also, I only tried the steps you described on alienfx-gui and I could have the keyboard lit up again after 

but if there are any specific steps you want me to reproduce with alienfx especially for the keyboard lighting issue, let me know!