T-Troll/alienfx-tools #524: Initial setup failed because path for udev is not really a path

Daniel100262, 2025-05-19:
**Describe the bug**
Initial setup failed because path for udev is a Buffered Reader and not really a path.

**To Reproduce**
Steps to reproduce the behavior:
1. Execute **setup.py**
2. Get the error:
```python
Traceback (most recent call last):
  File "/home/daniel/Downloads/alienfx/setup.py", line 89, in <module>
    shutil.copy(udev_file, udev_rules_dir)
  File "/usr/lib/python3.10/shutil.py", line 416, in copy
    dst = os.path.join(dst, os.path.basename(src))
  File "/usr/lib/python3.10/posixpath.py", line 142, in basename
    p = os.fspath(p)
TypeError: expected str, bytes or os.PathLike object, not BufferedReader

```

**Expected behavior**
To setup software correctly.

**System (please complete the following information):**
 - Model: AlienWare M15R7]
 - OS: Ubuntu 22.04

T-Troll, 2025-05-19:
Seems like you miss repository - i don't develop for Linux.

closed 2025-05-19 by Daniel100262

Daniel100262, 2025-05-19:
Sorry, wrong repository.