T-Troll/alienfx-tools #528: Unplugging or plugging in laptop causes color change, even when default profile has `Software Effects` turned off

AetherCollective, 2025-05-25:
**Describe the bug**

Everytime I unplug or plug in my laptop, the default profile is loaded and my colors change. How do I disable the colors changing when I unplug or plug in my laptop?

**To Reproduce**
1. Load a profile that isn't the default profile
2. Unplug or plug in the laptop
3. Watch as default profile loads and changes colors.

**Expected behavior**

With the `Software Effects` disabled for the default profile, I expect the colors to not change when the default profile is loaded.

**System (please complete the following information):**
 - Model: Alienware m17 R2
 - Model details: per-key rgb keyboard
 - OS: Windows 11 24h2

T-Troll, 2025-05-25:
- Turn off profile auto switch
- Make sure you don't have a separate default profile for DC (Profiles->Triggers, it can be switched in this case).

AetherCollective, 2025-05-25:
1. Profile auto switch is off
2. I only have one `Default` profile set as default. I'm unsure how you would get to this state as setting any profile as Default unsets it for the rest
3. Unsure what you mean by the last part. Do you mean I need both AC and Battery checked?
[image]
[image]
[image]

T-Troll, 2025-05-25:
> Unsure what you mean by the last part. Do you mean I need both AC and Battery checked?

Both checked or unchecked, it's the same.
But, being honest, seems like i found a glitch in power-driven profile selection. I'll change it into the next release.

AetherCollective, 2025-05-25:
I'll wait for the release and come back if I'm still having an issue. Thank you!