T-Troll/alienfx-tools #529: The ability to reorder your profiles

AetherCollective, 2025-05-25:
I'd like to be able to sort the list of profiles in the gui and tray menu, either via alphabetical or manual sort. Thank you

[image]
[image]

AetherCollective, 2025-05-25:
I just found the registry keys responsible for this sort, so I'm doing it manually. Still, would be nice to be able to have a GUI way to sorting 🙂

T-Troll, 2025-07-06:
Ohh... Miss this issue. Let me think, i'll do something.

T-Troll, 2025-07-06:
You see.... It's quite easy to make alphabetic sort, but not so easy to make custom sorting.

Seems like you have some special case about it (can you share it for me, please?), because initially profiles was designed for conditional auto-switch.

AetherCollective, 2025-08-30:
Hi, did this get missed again?

T-Troll, 2025-08-30:
Unfortunately, yes. It's quite hard to do for tray menu, i'll trying to configure out.

T-Troll, 2025-08-30:
Especially for you - this version has sort at Profiles tab.
[alienfx-gui.zip](https://github.com/user-attachments/files/22058324/alienfx-gui.zip)

AetherCollective, 2025-09-05:
Thanks man!

closed 2025-09-05 by AetherCollective