T-Troll/alienfx-tools #531: Run Action is buggy: sometimes launches another profile's action, sometimes does nothing

AetherCollective, 2025-05-25:
**Describe the bug**
The `Run Action` setting doesn't work very well. I think it's running the wrong Run Action from another profile, whichever one was last set.

**To Reproduce**
Set a Run Action on at least TWO profiles.
Switch between them. The older one stops working and runs the newer action.

Switch to said profile
watch as action does not trigger or wrong action triggers

**Expected behavior**
Switching to a profile should reliably run only its own action.

**Screenshots**
How about a short video? https://www.youtube.com/watch?v=melk6wvd4WE

**System (please complete the following information):**
 - Model: Alienware m17 r2
 - Model details: per-key rgb keyboard
 - OS: Windows 11 24h2

T-Troll, 2025-05-25:
I check now.... No, it's working as designed. Running app not stopped at profile switch.
But be careful, i use "Shell execution", so you can't pass command - exe, script, folder or document only.

In your case, seems like it not depend on my tools, but you desktop switcher app acting like this if multiply instance exist. You may need make a script for switch instead of running it directly.

AetherCollective, 2025-05-25:
You're right! So what was going on is my exe reads an ini file in the same directory as the script. My exe was using working directory, which is somehow influenced when ran with AlienFX-gui. I fixed my exe behavior by instead using the script directory