T-Troll/alienfx-tools #540: AlienFan-Gui system tray icon not visible

Enthu-dev, 2025-06-19:
Hi,

Win 11 pro 24H2, with Dell G15 5530 and 13650HX

I downloaded the zip bundle, as I only need to evaluate AlienFan-Gui.

While the application seems to change the AWCC Thermal profiles,
the system tray icon is invisible or blank.
Hovering mouse over that space shows power profile and fan RPM.
Hovering lightly outlines a whitish backround effect.

Using windows dark mode.

I have AWCC installed, fyi.

techlogik, 2025-06-19:
Came to report the same. Last several updates, not sure which, it is now just a blank space instead of the Icon in Green for Fan Control. Windows 11 Pro latest updates/patches.

T-Troll, 2025-06-20:
Got it. Under investigation.
It's strange for fan gui, icon is static there instead of fx gui.

T-Troll, 2025-06-20:
Confirmed and fixed, wait for the next release.

closed 2025-06-20 by Enthu-dev

techlogik, 2025-06-21:
Yep..thanks for the work you put into this...fixed!