T-Troll/alienfx-tools #544: Update button isn't doing anything

AetherCollective, 2025-06-25:
Got a notification for an update. I tried to click the update button but it does nothing. I had to click help to get to the github page to download the update.
[image]

T-Troll, 2025-06-25:
It's just update check. In case new version found, notification appears. Click on it.

T-Troll, 2025-06-25:
Check your notification settings, you can be in DnD mode or notification from the app not allowed.
It works well at mine systems.

AetherCollective, 2025-07-06:
I'm not in dnd mode and I get notifications when I start the app just not when I click the update button

AetherCollective, 2025-07-06:
nevermind you were right my notifications were turned off